speeduino/speeduino #1246: Test mode leaves ignition channels active when being stopped

ggurov, 2024-09-12:


noisymime, 2024-09-19:
The stop test mode button already calls `endCoilxCharge()` and `closeInjectorx()` on all channels when it's pressed so not sure how you're getting this.

What hardware are you testing on? I've tried turning the testing on/off 20-30 times on a mega2560 and can't reproduce the issue, but it's possible there may be some odd race condition occurring on other architectures.

ggurov, 2024-09-19:
it has to do with that waste spark mode in 6x3

```
    case TS_CMD_TEST_DSBL: // cmd is stop
      BIT_CLEAR(currentStatus.testOutputs, 1);
      endCoil1Charge();
      endCoil2Charge();
      endCoil3Charge();
      endCoil4Charge();
      #if IGN_CHANNELS >= 5
      endCoil5Charge();
      #endif
      #if IGN_CHANNELS >= 6
      endCoil6Charge();
      #endif
      #if IGN_CHANNELS >= 7
      endCoil7Charge();
      #endif
      #if IGN_CHANNELS >= 8
      endCoil8Charge();
      #endif
```

6x3 is possible when IGN_CHANNELS is = 3 , but there appears to be a special case handling waste spark in that mode, where it will actually turn on/off two channels with one timer, making it actually work correctly, as long as pinMode is set to output for the 5 + 6 coils. 

however, endCoil5Charge and endCoil6Charge never gets called if IGN_CHANNELS = 3

noisymime, 2024-09-19:
Ahh this is related to #1245 then. 6+3 on on a mega2560 was never intended to allow for wasted COP ignition given the 3 channels so these sorts of things are surprising. 

I'll close this for now and update #1245 accordingly

closed 2024-09-19 by noisymime

ggurov, 2024-09-19:
https://github.com/speeduino/speeduino/blob/master/speeduino/scheduledIO.cpp#L111-L117 it's that part that totally does 6 cyl wasted COP work.