speeduino/speeduino #1247: Flex Sensor Settings not in function

Bolino08C, 2024-09-14:
Hello, in the firmware. I see a fixed value of flexCounter. (timer.cpp)

 if(configPage2.flexEnabled == true)
    {
      byte tempEthPct = 0; 
      if(flexCounter < 50)
      {
        tempEthPct = 0; //Standard GM Continental sensor reads from 50Hz (0 ethanol) to 150Hz (Pure ethanol). Subtracting 50 from the frequency therefore gives the ethanol percentage.
        flexCounter = 0;
      }
      else if (flexCounter > 151) //1 pulse buffer
      {

        if(flexCounter < 169)
        {
          tempEthPct = 100;
          flexCounter = 0;
        }
        else
        {
          //This indicates an error condition. Spec of the sensor is that errors are above 170Hz)
          tempEthPct = 0;
          flexCounter = 0;
        }
      }
      else
      {
        tempEthPct = flexCounter - 50; //Standard GM Continental sensor reads from 50Hz (0 ethanol) to 150Hz (Pure ethanol). Subtracting 50 from the frequency therefore gives the ethanol percentage.
        flexCounter = 0;
      }
==================================================
I modded like this, but I don't really know how to deal with error conditions  :
--------------------------------------------------------------------------------------

if(configPage2.flexEnabled == true)
    {
      byte tempEthPct = 0; 
      if(flexCounter < configPage2.flexFreqLow ) // Modded by Bolino08C  (84 for chinese sensor)
      {
        tempEthPct = 0; //Standard GM Continental sensor reads from 50Hz (0 ethanol) to 150Hz (Pure ethanol). Subtracting 50 from the frequency therefore gives the ethanol percentage.
        flexCounter = 0;
      }
      else if (flexCounter > (configPage2.flexFreqHigh +1) ) //1 pulse buffer 150 +1 orig Modded by Bolino08C (184 + 1 for chinese sensor)
      {

        if(flexCounter < 203) // 169 orig Modded by Bolino08C for chinese sensor
        {
          tempEthPct = 100;
          flexCounter = 0;
        }
        else
        {
          //This indicates an error condition. Spec of the sensor is that errors are above 170Hz)
         // tempEthPct = 0;   Modded by Bolino08C for chinese sensor
         // flexCounter = 0;  Modded by Bolino08C for chinese sensor
        }
      }
      else
      {
        // tempEthPct = flexCounter - 50; //Bolino08C Standard GM Continental sensor reads from 50Hz (0 ethanol) to 150Hz (Pure ethanol). Subtracting 50 from the frequency therefore gives the ethanol percentage.
        tempEthPct = flexCounter - configPage2.flexFreqLow; //Standard GM Continental sensor reads from 50Hz (84 for Chinese Sensor) (0 ethanol) to 150Hz (Pure ethanol). Subtracting 50 from the frequency therefore gives the ethanol percentage.
        flexCounter = 0;
      }

noisymime, 2024-09-15:
Do you have any details of the sensor you're using? I've never seen any flex sensor that doesn't use the same 50-150Hz range as the GM one is all, even the knockoff / Chinese sensors.

Bolino08C, 2024-09-16:
Here it is. 
Passat B7 CC 4Motion Santana 06H907811A 06H 907 811 A
https://a.aliexpress.com/_EuD0GsV
When I bought it, people said it doesn't work when it's the frequency that's not good.

noisymime, 2024-09-19:
I'm a little doubtful that the sensor you have is actually meant to be running at anything other than 50-150, but I do agree that those settings should be used within the code. 
I've pushed a change up now to address this.

closed 2024-09-19 by noisymime

Bolino08C, 2024-09-20:
Thanks a lot !

commit c2512bed (2024-09-19): Use flex fuel frequency settings for non-GM sensors. Closes #1247