speeduino/speeduino #1253: 8 Squirts per engine cycle bug causing Zero duty cycle, lots of other problems

youngfg, 2024-10-04:
I have a NO2C board running 2024.02.2 firmware, on a simulator.
I have it setup for an 8 cylinder with a distributor and TBI with two injectors simultaneous.
When I setup 4 squirts per engine cycle everything looks fine in tunerstudio , pulse width goes up with the MAP, duty cycle goes up with RPM.
For a dual plane TBI V8 engine, 8 squirts per engine cycle seems like the best setting. I think you would want to inject fuel for each cylinder as it fires.

When I change to 8 squirts per engine cycle, the duty cycle in tunerstudio stays a 0.0% no matter what you do. The injector outputs do pulse for every cylinder but don't change PW. Also the TS gage nSquirts shows 0 squirts.

I erased the eeprom reloaded 202310 firmware. made a new TS project, loaded the base tune, and only made changes to the engine constants , and trigger setup.
Still doesn't work right with 8 squirts. Switch back to 4 squirts works fine, switch to 8 stops working.
Everything looks right with 4 squirts.
I have noticed lots of things wrong with 8 squirts. you can see in the attached logs and tunes.
FYI 4 cylinder and 4 squirts works right.

Attached tune and log files
[Documents.zip](https://github.com/user-attachments/files/17258458/Documents.zip)

Batt voltage is doubled,
ignition load is zero
fuel load is zero
Baro is doubled
duty cycle is zero
loops per rev is zero
advance is zero
Probably a lot more errors I am not seeing.

MykkHan928, 2024-10-07:
My first suggestion would be to retest using the Master firmware to have all of the most recent updates and changes made to the firmware. If I recall some crank trigger work has recently been done in the basic distributor code
Taking a look at your 8 squirt tune and datalog. There appears to be some currupt data somewhere in your set up, despite the TPS remaining 0% the TPDdot is going absolutely haywire which is not normal. 

See if adding some trigger angle offset changes the stuck ignition timing value, I believe most aftermarket efi set ups that utilize a distributor need some sort of offset to detract timing values from.

youngfg, 2024-10-08:
I found the Master version on Speedyloader, cleared eeprom, loaded, loaded Master firmaware, used Master INI file, loaded base tune. Only changed engine constants, and trigger setup. Does the same thing.
[master rfirmware log files.zip](https://github.com/user-attachments/files/17294098/master.rfirmware.log.files.zip)
I ran it on the simulator with four squirts, downloaded the log and tune, changed to eight squirts downloaded the log and tune

youngfg, 2024-10-09:
How can you have 8 squirts if you only have three bits allocated for the number of squirts?

TS ini file  
nSquirts          = bits,     U08,       84, [5:7]

Globals.h
#define BIT_STATUS3_NSQUIRTS1     5
#define BIT_STATUS3_NSQUIRTS2     6
#define BIT_STATUS3_NSQUIRTS3     7

youngfg, 2024-10-13:
Found the problem above, evidently someone else has been down this road before.
In the TS.ini there was a line commented out that used ncylinders/divsor to calulate nsquirts.
Used that and all is well now

closed 2024-10-13 by youngfg