speeduino/speeduino #1257: Missing tooth decoder works with single tooth cam, but code says it doesn't

clytle374, 2024-10-16:
Maybe I'm really making a mistake here, but it seems that the "missing tooth" setting in tuner studio works with a crank wheel and single tooth cam sensor.  This is "missingtooth" in the speeduino code, I'm pretty sure it is anyway. But the following code comment above missing tooth has me confused.  What am I misunderstanding?  

  
`/** A (single) multi-tooth wheel with one of more 'missing' teeth.
* The first tooth after the missing one is considered number 1 and is the basis for the trigger angle.
* Note: This decoder does not currently support dual wheel (ie missing tooth + single tooth on cam).
* @defgroup dec_miss Missing tooth wheel
* @{`

MykkHan928, 2024-10-16:
Further down the code on line 519 it has checks if sequential is chosen and then it looks for the secondary trigger. The original note is probably left over from using only a crank speed missing tooth trigger with no secondary for paired cylinder and waste spark.

clytle374, 2024-10-17:
I kinda assumed as much, just wanted confirmation and a note it can be deleted.

commit 4e23e87a (2024-11-07): Correct comment. Fixes #1257
closed 2024-11-07 by noisymime