speeduino/speeduino #1258: SD card logger can stumble the engine

drozdowsky, 2024-10-19:
This was already touched in speeduino discord [here](https://discord.com/channels/879495735912071269/937553974905675816/1294617091156082709), posting this here as it can get lost.

Board: BMW PnP by Pazi + STM32_Mega (this could potentially only happen on STM32).
release: 2023.05

Basically one/twice per ride (10-20 minutes) it feels like engine skips a cycle, disabling SD card logger fixes it.

I attach the log here, the 267.075 timeframe captures the stumble, basically a lot of values there are shifted (which results in bizzarre values for most of fields), also it slowed down below 10Hz set logging frequency. To me it seems like some operation with SD card took longer than anticipated and then the stumble happened.
[SPD_0015.csv](https://github.com/user-attachments/files/17444453/SPD_0015.csv)

noisymime, 2024-10-20:
Yeah you can see on 267.075 that is hasn't finished the write for that row and hence the following record (265.175) starts on the same row rather than the next one. 

The stm32 SD logging uses a very different mechanism to the Teensy as it doesn't have the same buffering system available in hardware. I'll take a closer look and see if I can reproduce it on an stm32. re you seeing this repeatedly or just occasionally?

drozdowsky, 2024-10-20:
Something in between, I had it almost on almost every 10-20 minute ride (I had about 5 of them) - as mentioned it usually only occurred once per ride. I haven't tried different SD card as I didn't have another one but I could try some faster SD card in the future also

noisymime, 2024-10-21:
OK, fairly sure I can see what is happening here. Looking at the log you provided, the first instance of the issue occurs after around 1248 bytes, which is very close to the ringbuffer size of 1270 bytes. 

What I believe is happening is that the ringbuffer is filling faster than the stm32 is able to write the data to the card, at least under some conditions (in particular the card itself). 

There's 2 things I'll try here, increasing the size of the ringbuffer, but also preventing further entries from being added to the buffer if if's more than a certain amount full. 

If I do a separate branch, are you able to compile and try an dev firmware?

drozdowsky, 2024-10-21:
@noisymime unfortunately currently I am abroad for 3 months so that will have to wait but sure after I come back I could do that

noisymime, 2024-10-22:
No stress, I managed to reproduce this with an stm32 and have pushed up some changes now that I'm 99% sure fix this.

commit 8b97b980 (2024-10-22): Increase SD ringbuffer size and add checks for space before writing to ringbuffer

Fixes #1258
closed 2024-10-22 by noisymime