speeduino/speeduino #1259: Feature request: Idle valve anti-stall

Nikolai521, 2024-10-20:
We need a way to prevent fast rpm drops, ones that are too fast for the PID to react at time, this is very noticiable on parking or driving a pretty low rpms

After talking with a friend and looking some OEM ecus, we noticed that when RPM falls from x point (ie 700rpm) the IAC valve shoots very fast to a specific value, the mine is about 86steps for preventing the engine for kill itself, also this can help with fast clutch drops

For now, i programmed a output to idle-up pin doing this, worked pretty well but it needs some tampering time to target for prevent oscillations but it helps for prevent stalls (for now) but its quite annoy

Soo this is my point
Possiblity of set a limit value of low RPM, below this the valve opens to a fixed **added** value (same as idle-up), the the user can set a tampering time to normal target steps or pid steps after x seconds (basically just going back to normal pid)

IM NOT a programmer, is just a idea.

Admanirv123, 2024-10-24:
Have you tried ignition advance too in idle advance settings? It'll react quicker.

Nikolai521, 2024-10-25:
> Have you tried ignition advance too in idle advance settings? It'll react quicker.

Yep, we tried that, its almost mandatory but barely can hold the lights load, probably the advance works better on bigger engines, mine is a 1.4l peugeot, my friend have a 1.1 fiat

On mine the stepper jumps from 39 to 57 steps only with lights, on clutch dumps is worse and the pid is not fast enough

ponzano, 2024-12-09:
> Yep, we tried that, its almost mandatory but barely can hold the lights load, probably the advance works better on bigger engines, mine is a 1.4l peugeot, my friend have a 1.1 fiat
>
>
>
> On mine the [...]

Is the 1.1 fiat multipoint injection? I have a 1.0 fire with multipoint and got idle pretty well sorted, if it uses the same intake I could share my settings

closed 2024-12-09 by Nikolai521