speeduino/speeduino #1264: Incorrect pin listing in /reference/speeduino.ini

OpenLogicEFI, 2024-11-17:
Located in the current /reference/speeduino.ini

Looks like oil/fuel pressure have pins defined that don't exists. Line 1218 & 1220

```
oilPressurePin      = bits,   U08,     135, [3:7], "A0", "A1", "A2", "A3", "A4", "A5", "A6", "A7", "A8", "A9", "A10", "A11", "A12", "A13", "A14", "A15", "A16", "A17", "A18", "A19", "A20", "A21", "A22", INVALID, INVALID, INVALID, INVALID, INVALID, INVALID, INVALID, INVALID, INVALID`

fuelPressurePin     = bits,   U08,     136, [0:4], "A0", "A1", "A2", "A3", "A4", "A5", "A6", "A7", "A8", "A9", "A10", "A11", "A12", "A13", "A14", "A15", "A16", "A17", "A18", "A19", "A20", "A21", "A22", INVALID, INVALID, INVALID, INVALID, INVALID, INVALID, INVALID, INVALID, INVALID
```
I think it should be:
```
oilPressurePin      = bits,   U08,     135, [3:6], "A0", "A1", "A2", "A3", "A4", "A5", "A6", "A7", "A8", "A9", "A10", "A11", "A12", "A13", "A14", "A15", 

fuelPressurePin     = bits,   U08,     136, [0:3], "A0", "A1", "A2", "A3", "A4", "A5", "A6", "A7", "A8", "A9", "A10", "A11", "A12", "A13", "A14", "A15", 
```

-Detonation

noisymime, 2024-11-17:
The A16-A22 pins are valid for Teensy boards, which is why they're listed in the ini file. 

These could potentially be wrapped in a check for which board is in use to avoid confusion though I suppose.

OpenLogicEFI, 2024-11-17:
Ohh got it. That makes sense. As we discussed a few weeks ago I was working on a ini adding a stm32 raw pin selection when i noticed it. I have some small things to work out, double check my work ect.. but have a functional idea going. I can add the Teensy if you want. Although i don't think this specific situation will cause much confusion, just something i noticed. 

what I have so far. (unfinished, please excuse unpolished areas)
https://github.com/OpenLogicEFI/speeduino/blob/master/reference/speeduino.ini