speeduino/speeduino #1265: TunerStudio MS msq files does not contain ECT/IAT/VSS calibration

AlexE75, 2024-11-18:
Hello,

I noticed long ago that when swapping mega2560 board some values are not transferred when loading saved msq file via TunerStudio. I need to recalibrate:
- ECT
- IAT
- VSS and gears

not sure about o2 sensor

kind regards
Aleksander

Autohome2, 2024-11-18:
This is not an issue, it is the way tunerstudio works unfortunately.
When you use the calibration function in TS it doesn't send that data to the speeduino, it sends the calculated steps based on the points you entered .
TS cannot reconstruct the calibration points from that info.

AlexE75, 2024-11-19:
I understand it conforms to ECT IAT o2 calibration.

How about VSS and gear parameters? They also ave to be restored manually

kind regards
Aleksander