speeduino/speeduino #1270: Secondary Spark table. Switched - Input based not working

17Blue, 2024-12-06:
The secondary spark Table switching does not work using input based. 
[image]
Using the same pin i can get the secondary Fuel table working using input based (had to use external pull up resistor - the internal pull up resistor does not work).
I noticed that the I_O summary on Tuner Studio does not list the Spark2 input pin.
 
[image]
I am using the latest firmware Feb 2024 release with February 2024 Fix Pack 2.

noisymime, 2024-12-06:
Can you pop up a log and your tune? I just gave it a quick try and it switched over OK on the bench.

I'll run  up your tune though and see if I can get the same behaviour. 

Will also make sure to add the pin to the IO summary as well

17Blue, 2024-12-06:
[CurrentTune.zip](https://github.com/user-attachments/files/18037204/CurrentTune.zip)
Attached is the tune.  Will upload the log once I have logged it.
Thanks.

noisymime, 2024-12-10:
I just tried your tune and it switches over fine. I did not that the Secondary Spark table mode was set to off in the tune you posted, but I simply changed that to `Switched - Input based`, set the polarity to `LOW` and enabled the internal pullup. Then applying ground to pin 30, the ignition switched between the 2 tables and the `spark2Active` flag is being set. 

I'll close this for now as it appears to be a tune and/or hardware issue.

closed 2024-12-10 by noisymime