speeduino/speeduino #1283: Instantaneous MAP mode should not filter "invalid" readings?

adbancroft, 2025-01-26:
#1157 applied existing MAP ADC filtering (`(tempReading < VALID_MAP_MAX) && (tempReading > VALID_MAP_MIN)`) uniformly across all MAP sampling modes.

This is a change in behavior: prior to#1157, the instantaneous sampling mode did *not* skip readings outside the valid range (it set a global error flag - which was never used).

noisymime, 2025-05-04:
Is this actually an issue? 

I understand that it's a change in behaviour for the instantaneous mode, but I think what was introduced in #1157 is really the desired result. Have you seen any issues with the code from 1175?

adbancroft, 2026-02-21:
Closing as the change in behavior doesn't seem to have affected anyone.