speeduino/speeduino #1284: Irradic Flex Fuel readings on Teensy 4.1

RempageR1, 2025-01-27:
I was testing a generic Flex-Fuel sensor (#13577429) on the Teensy 4.1 using E5 petrol but got very irradic readings. To start, I connected my scope to the pin directly on the Teensy:

[image]

And that looked clean enough :).

Looking at the code, the FlexFuel sensor input gets an interrupt attached that triggers plexPulse() on change. If the pin then reads true, flexCounter is increased by 1. After a second, the pulses are counted and the frequency is known. However when adding some simple debug code, the READ_FLEX() dit not return true, false, true, false etc. but true, true, false, true, false, false etc. It seems the Teensy really needs a VR conditioner or so to read such values. So I added a check to flexPulse(void) to check if the last value was different. That got it 99% working, but still some irradic spikes. 

This was caused by some pulses being so quick they could never be correct, like the second pulse after 20 microseconds. To fix this, I only allowed pulses that had a delta > 1ms. So if a sensor stays below 1000hz, it should be fine :).

The final code looks like this which outputs 55hz, exactly matching my E5 petrol: 
------------------------------------------
void flexPulse(void)
{
  if(READ_FLEX() == true)
  {
    if (lastFlex == false){
      if ((micros() - flexStartTime) > 1000){
        uint16_t tempPW = clamp(micros() - flexStartTime, 0UL, (unsigned long)UINT16_MAX); //Calculate the pulse width
        flexPulseWidth = LOW_PASS_FILTER(tempPW, configPage4.FILTER_FLEX, flexPulseWidth);
        ++flexCounter;
      }
      lastFlex = true;
    }
  }
  else
  {
    if (lastFlex == true){
      flexStartTime = micros(); //Start pulse width measurement.
      lastFlex = false;
    } 
  }
}
------------------------------------------

I think this will also help other users with the Teensy. If needed, I can clean up the code a bit and create a pull request. Just let me know. Thank you.

noisymime, 2025-01-27:
Can you try adding this code into the `setTriggerHysteresis()` function within the `board_teensy41.cpp` file and see if it makes any difference?

```
//Flex trigger
p = digital_pin_to_info_PGM + pinFlex;
*(p->reg + 1) &= ~(p->mask); // TODO: atomic
*(p->pad) = padConfig;
*(p->mux) = 5 | 0x10;
```

RempageR1, 2025-01-29:
Hi noisymime,

I`ve tested that and indeed that fixes the problem and is also a much nicer solution :)

Thank you!

noisymime, 2025-01-29:
@RempageR1 Thanks for confirming! A fix has been pushed for this now that also includes the VSS and knock external interrupt pins.

commit e56c6f0a (2025-01-29): Add Teensy 41 pin hysteresis on all external interrupt functions (Flex, VSS and Knock)

Fixes #1284
closed 2025-01-29 by noisymime