speeduino/speeduino #1293: Miata VSS is not detecting in VSS tab

MrZebraaa, 2025-04-12:
Hello

I do not get any VSS on my SpeedyEFI 01-05 Miata PnP ECU. I am on 202501. I have been dealing with this for a little bit now. I know that my issue isnt hardware related on the vehicle. The pin on the ecu harness where the VSS signal is output from the speedometer is showing 5V pulses. My speedometer, odometer, and cruise control work as well. I have tried every option on TS and I do not get any speed reading out of any of them. constantly showing 0. When I have an aux input watching pin 2 i do see pulses, however TS does not register any of these. 60kmh auto calibrate does nothing, setting the smoothing to minimum or maximum does nothing. Is it in the firmware by chance? ive been looking through it and have not been able to make out anything out of the ordinary besides vss_uses_rpm2. Does that do anything that i should be concerned about?

[image]

noisymime, 2025-05-21:
Have you tried manually setting the VSS Pin setting to 2? I believe 2 is the correct pin on that board, but it would be good to confirm. 

By the looks of it there was no default value in place for VSS on that board. I'll get that updated to 2 if it can be confirmed to be the correct one.

MrZebraaa, 2025-09-29:
Hello, sorry for the massive delay.

I have confirmed pin2 is the correct pin for vss on the board. in the latest fw, i have pinvss set as 2

[image]

It still does not detect at all in TS, not even an auxin set to watch pin2 sees it. I know my vss works as backprobing the pin on the ecu harness gives me 0-5v pulses. The cruise control also works in my car as well and that relies on VSS. I have replaced both the adapterboard and the arduino. When i was still focused on this issue, i noticed that sending 0-5v pulses with a different arduino to that same pin on the bar arduino(no adapterboard) refused to detect them either. What do you think this issue could be?