speeduino/speeduino #1314: UA4C/Ocelot board define pinVSS

itszeekay, 2025-07-16:
With the Ocelot the UA4C board definition is used, however it includes additional input for VSS. This is wired to pin 2. Since pinVSS isn't defined in the ua4c board, it defaults to pin 6. However TS does not allow the selection of pin 2 for the purpose.

[image]

Setting pin 2 as the default should allow 'board default' to work as expected.

This was confirmed by turboedge in discord https://discord.com/channels/879495735912071269/937475344926146620/1386812296470204438

NSS-Built, 2025-08-25:
YES!!!!! Need this ASAP. Currently running a GPS for VSS but for our local laws we need it to be driven off the gearbox so need VSS Pin sorted asap!!!

closed 2025-09-01 by noisymime

NSS-Built, 2025-09-01:
Cheers Josh,

Still very new to all of this so learning as i go

commit 6f908100 (2025-09-01): Add default VSS pin for UA4C. Fixes #1314