speeduino/speeduino #1317: MISRA scan is broken

adbancroft, 2025-09-08:
Commit https://github.com/speeduino/speeduino/commit/0fb84fe808009eb0e48e5899b153f1719b512186 has broken the MISRA scan:
```
        <error id="preprocessorErrorDirective" severity="error" msg="failed to evaluate #if condition" verbose="failed to evaluate #if condition">
            <location file="C:\dev\speeduino\speeduino\statuses.h" line="13" column="0"/>
        </error>
```

noisymime, 2025-09-10:
This is looking a bit messy. Using a more up to date version of cppcheck (2.13) makes this error go away, but for some reason it's producing significantly less findings that the older version. Many files either aren't being scanned or aren't generating the same messages that they were in the earlier cppcheck version. 

I'm thinking cppcheck may need to have access to some of the standard header files, even though it claims it doesn't. There's a lot of errors in the results files that look like they're related to missing typedefs and things. 

Might be worth investigating the Sonar solution more as I don't think putting bandaids on top of bandaids with cppcheck is worth it.

adbancroft, 2025-09-16:
> for some reason it's producing significantly less findings that the older version.
> I'm thinking cppcheck may need to have access to some of the standard header files,

I doubt it's the headers. cppcheck started using a [split licensing model (OSS + commercial)](https://www.cppcheck.com/plans-pricing) a few versions ago - probably that.

adbancroft, 2025-09-16:
[results.zip](https://github.com/user-attachments/files/22371461/results.zip)

I ran the CodeQL MISRA queries - lots of issues found! Results attached: use the VS Code Sarif viewing extension to visualize.

adbancroft, 2025-09-16:
To run the CodeQL MISRA scan I had to use the library PR (#1306) - the scan stalled otherwise. My suspicion is that the libdivide headers generated too large a search space for some queries (specifically `constant_fast_div.h/hpp`)